PX4/PX4-Bootloader #27: Bootloader will not jump to app

ksschwabe, 2015-03-27:
There is a problem in the current bootloader logic (at least on the Px4FMU board) where it will not boot. I have found the problem and put in a fix in my pull request for parallel USB and USART bootload ability. If you don't merge that in soon, perhaps just make patch based on my comments below and push that so that if someone does update their bootloader off the master branch they don't have a problem.

The problem is in cdcam.c and looks as if it was introduced on 2nd February 2015. The offending line of code is this:
static void cdc_disconnect(void)
{
    usbd_disconnect(usbd_dev, true);
    usbd_dev = NULL;
}

Previously, it just called usbd_disconnect(true);

The problem is that this cdc_disconnect(void) function is called in cfini which is called in jump_to_app(). When jump_to_app() is called in the if(try_boot) section of code, the usbd_dev has not yet been assigned. It is assigned only by cinit().

To fix this error I have just moved the following code to be above the if(try_boot) code section.

 /* configure the clock for bootloader activity */
  rcc_clock_setup_hse_3v3(&clock_setup);

cinit(BOARD_INTERFACE_CONFIG_USB);

davids5, 2015-03-27:
1) Perhaps the change was Feb if 2014?
2) Should we consider not reordering the code and use something like this?
```
static void cdc_disconnect(void)
{
       if(usbd_dev) {
	   usbd_disconnect(usbd_dev, true);
	   usbd_dev = NULL;
        }
}
```

ksschwabe, 2015-03-27:
@davids5 
1) Yes, I think you are correct. The actual change was made in Feb 2014 (not Feb 2015 as I said - I think I misread), but it looks as if it was only committed into the project by you on 16 Dec 2014, Hence why I my previous version of playing around with the bootloader in November last year was still working.
2) Your solution is certainly more elegant than mine. I have tested it and it seems to work. If you can make the change and confirm that it works for you as well, then we can push the patch.

davids5, 2015-03-27:
Thank you for reporting this Bug and the work that you have done on the dual boot. I have built, tested pushed the critical bug fix. 

FYI: @tridge, @LorenzMeier - this my explain a lot

Since the dual boot changes are substantial, we will need time to review them thoroughly .

Thank you again for all your efforts!

closed 2015-04-02 by davids5