PX4/PX4-Bootloader #29: the problem for ""USB device not recognized" ,is the problem jum to app?

molyswu, 2015-04-01:
hello ,davids5 !
I used STLINK-V2 full chip erase px4flow boarder and  build px4  Bootloader file (2015-03-27)
, make all for px4flow px4flow-bl.bin , build succed and connect STLINK-V2 ,then program the px4flow boarder in Bootloader(c/px4/Bootloader).restart the boarder  show "USB device not recognized" .
didn't communicated with PC USB, is the problem jum to app?

Thank you very much for your help!
molys
Email:sasunwu@126.com
2015-4-1

molyswu, 2015-04-01:
cdc code :
static void cdc_disconnect(void)
{
	        if (usbd_dev) {
            usbd_disconnect(usbd_dev, true);
            usbd_dev = NULL;
        }
}

--------------------------------------------
main_f4   code:

------------------------
------------------------

if (try_boot) {

#ifdef BOARD_BOOT_FAIL_DETECT
		board_set_rtc_signature(BOOT_RTC_SIGNATURE);
#endif

	jump_to_app();

		timeout = 0;
	}

	
	rcc_clock_setup_hse_3v3(&clock_setup);

	
	cinit(BOARD_INTERFACE_CONFIG);

davids5, 2015-04-01:
@molyswu I am not familiar with the px4flow board. But I would assume you can debug it with gdb using your Jtag. 

See 
https://pixhawk.org/dev/px4_bootloader
https://pixhawk.org/dev/gdb_cheatsheet
https://pixhawk.org/dev/jtag/start

molyswu, 2015-04-02:
Checking STM32F407VG  "clock_setup "  with GPIO 12PIN ,13PIN ，voltage only have 0.7V. Righten   voltage have 1.7V on tow PIN。boot cann‘t jump to app.

molyswu, 2015-04-02:
I was under the windows compiler and debug PX4 Toolchain /eclipse/clonse .

closed 2015-04-02 by molyswu