PX4/PX4-Bootloader #30: vbus pin configuration disagree in bootloader and px4firmware

KIrill-ka, 2015-04-24:
Bootloader code configures vbus pin with pulldown enabled.

    #ifdef INTERFACE_USB
    /* enable GPIO9 with a pulldown to sniff VBUS */
    rcc_peripheral_enable_clock(&RCC_AHB1ENR, RCC_AHB1ENR_IOPAEN);
    gpio_mode_setup(GPIOA, GPIO_MODE_INPUT, GPIO_PUPD_PULLDOWN, GPIO9);
    #endif

According to stmf40x manual pullup/pulldowns must not be enabled for 5v pins.
6. To sustain a voltage higher than V DD +0.3, the internal pull-up and pull-down resistors must be disabled.
The documentation mentions  1k1 pulldown for VBUS (which obviously has to handle 5v), but it's may be totally separate from normal GPIO pulldown.

In contrast to that, PX4Firmware doesn't enable pulldown. OTGFS vbus sensing configuration is also different. I see OTGFS_GCCFG_NOVBUSSENS bit set in px4firmware. OTG_FS_GCCFG_NOVBUSSENS is not set in libopencm3.

budebulai, 2017-01-07:
Pixhawk flashed latest bootloader, when USB plugged, pixhawk starts very well, but it fails to start with usb unplugged.

davids5, 2017-01-08:
@budebulai FYI @LorenzMeier 

I just built and tested Bootloader on master 38288b9912749812b9d27d0615c5025d81e6e787 and it is working fine on a 2.4.6 HW. 

The symptom you are describing was fixed in https://github.com/PX4/Bootloader/commit/b75a627a23e97596566faf31044bff1a494b0ec6 

Please check the git hash you are building and ensure the device is being reprogrammed with the bin file from 38288b9912749812b9d27d0615c5025d81e6e787

davids5, 2017-01-09:
Closing this - @budebulai please re-open if you have a confirmed problem with current master.

closed 2017-01-09 by davids5