PX4/PX4-Bootloader #33: firmware update via UART

magicrub, 2015-07-31:
Add ability to perform Pixhawk firmware update via UART. I hear this is possible already with a UART-only build version but we need a USB+Serial single image. This assumes it's possible to squeeze both into the constrained size.

ksschwabe, 2015-09-01:
It is possible to have  USB+USART bootlaoder in a single image. I have opened a pull request that has this ability, but it still has to be validated and merged in to the Px4 master repository.

If you want the ability to do firmware updates via the UART port so that you can do the update over a wireless link, be warned that the bootloader currently does not cater for bit and byte drops, which are likely to occur over a wireless link. This means that currently if a byte drop occurs during a firmware upload, the upload will simply fail and you will have to start the upload again.

If, however, you are using the USART booloader with a cable, it should work without problems.

magicrub, 2015-09-01:
Thank you @ksschwabe !

magicrub, 2015-11-11:
Merged! https://github.com/PX4/Bootloader/pull/36

closed 2015-11-11 by magicrub