PX4/PX4-Bootloader #38: Option "flto" the GCC.

rydsfoz, 2015-10-07:
Hello everyone,

some of the supported platforms are relatively close to their maximum sizes, for example, PX4_PIO_V2 with 3484 bytes (85% of 4096). If been added new features this can become a problem. One way of reducing the program size is to use the option "flto" the GCC.

Below follows some results:

                 | normal | nano | flto | nano-flto | dif % |
-----------------|--------|------|------|-----------|-------|
PX4_DISCOVERY_V1 |  7908  | 7548 | 7024 |    6660   | 17.9  |
PX4_PIO_V2       |  3348  | 3348 | 2468 |    2468   | 29.2  |

personal project

        | p1-avr | p2-stm32f1 | p3-stm32f4 |
--------|--------|------------|------------|
Os      |  23911 |    34180   |    75384   |
Os-flto |  21015 |    11552   |    33036   |
O3      |  27219 |    43708   |    93416   |
O3-flto |  29813 |    13416   |    41460   |

This option "flto" could be used in the future if the sizes of the programs become a problem.

davids5, 2015-10-07:
What version of GCC are you running?

rydsfoz, 2015-10-07:
In windows 4.9.3.

davids5, 2015-10-07:
Please run gcc --version and report that back here,

rydsfoz, 2015-10-07:
arm-none-eabi-gcc --version

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rydsfoz, 2015-10-07:
In VM (Debian 8)

arm-none-eabi-gcc --version

arm-none-eabi-gcc (4.8.4-1+11-1) 4.8.4 20141219 (release)
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rydsfoz, 2015-10-07:
In windows: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release)
In In VM (Debian 8): arm-none-eabi-gcc (4.8.4-1+11-1) 4.8.4 20141219 (release)

The tests that were done on windows.

davids5, 2015-10-08:
What are the results on the Linux box?

rydsfoz, 2015-10-08:
In linux without changing the code, only changing the compiler options:

                 | normal | nano-flto |dif %|
-----------------|--------|-----------|-----|
px4aerocore_bl   | 8292   | 7016      |15.4 |
px4discovery_bl  | 8116   | 6808      |16.1 |
px4flow_bl       | 8096   | 6792      |16.1 |
px4fmu_bl        | 8332   | 7072      |15.1 |
px4fmuv2_bl      | 8472   | 7228      |14.7 |
px4io_bl         | 3484   | 2528      |27.4 |
px4mavstation_bl | 7596   | 6388      |15.9 |

The library "libopencm3" was compiled with the same parameters
and used the "gcc-ar" program.

davids5, 2015-10-08:
@rydsfoz 

Sorry to ask so many questions, but we had run into a series of issues in the past with lto and certain optimizations setting. One was that gbd could not load the code, other issues were the code  would not run properly. 

1) What changes did you make to the build? (diff or PR would be helpful)
2) Did you load and run the builds with gdb?
3) From the list above, which builds did you test?

rydsfoz, 2015-10-09:
So far the program runs properly with the LTO option, but as this is a critical program is necessary caution, many tests would have to be made for adopting new options.You'd better leave the defaults for now since they work very well.

1) The changes I made to compile with the LTO option will be sent as PR.

2) I did it today. I used GDB to track program execution, worked but had to indicate the path of the library code "libopencm3".
In the document "Using the GNU Compiler Collection" mentions:
"Link-time optimization does not work well with generation of debugging information.
Combining ‘-flto’ with ‘-g’ is currently experimental and expected to produce unexpected results."

3) I tested "PX4_DISCOVERY_V1" (I do not know if i understand the question).

rydsfoz, 2015-10-09:
The command I use to gdb:
arm-none-eabi-gdb px4discovery_bl.elf -ix teste.txt --directory=libopencm3/lib/stm32/f4

contents of file test.txt
target remote localhost:4242

LorenzMeier, 2021-01-01:
Closing as stale.

closed 2021-01-01 by LorenzMeier