PX4/PX4-Bootloader #43: Add Pixhawk+ ID to boot loader and prevent boots if silicon rev < 3 on the FS peripheral

LorenzMeier, 2015-11-08:
This might also be the right occasion to register an USB ID ourselves. The goal is to make "fixed" hardware easily identifiable and allow the use of the full flash space for new boards without jumping through hoops.

ksschwabe, 2015-11-09:
@LorenzMeier : Don't you want to do this rather in the px_uploader.py script? The chip ID is already returned by the bootloader to the uploader script, so you will be able to figure out the revision of the silicon, and then get the uploader to tell you whether you can use all 2MB of the flash or only the one bank of 1MB.

LorenzMeier, 2015-11-09:
Yes and no. As a customer you want to know what you're buying. If we don't enforce the silicon revision in our process then manufacturers will have no workflow to ensure they are offering silicon rev. 3. If we fail the boot once they flash the new boot loader to old silicon we can make sure that when they start to offer a Pixhawk + it is indeed a +.

ksschwabe, 2015-11-09:
Ok. Understood.

LorenzMeier, 2015-11-26:
@davids5 Once you're back this is reasonably high priority. We need: 

* A bootloader command to query silicon revision
* The option to fail the boot loading by emitting a specific failure code if the wrong silicon is present. This has to be in the boot loader.

davids5, 2015-12-01:
@LorenzMeier 

I need some clarification 

First, am I correct in assuming that since this is a protocol level change - we will have to bump  the bl_proto_rev to 5 (The value returned by the PROTO_DEVICE_BL_REV command) if we add a new command or error code?

* A bootloader command to query silicon revision

   We have the command PROTO_GET_CHIP it returns the 32 bits per:
[image]

Do you want an additional command PROTO_GET_SILICON_VERSION?

Would it return the decoded ASCII code of A, Z , Y , 1 or 3 from the value in bits 31:16 or just the binary value of bits 31:16 from  DBGMCU_IDCODE?

* The option to fail the boot loading by emitting a specific failure code if the wrong silicon is present. This has to be in the boot loader.

When you say "option" - are you referring to a compile time option?

If it is a compile time option, then when the F4 variants are built with it on, we will issue a new "Error Code" in response to just PROTO_CHIP_ERASE and PROTO_PROG_MULTI?

Since the current return codes are none error specific
```
#define PROTO_OK		0x10    // INSYNC/OK      - 'ok' response
#define PROTO_FAILED		0x11    // INSYNC/FAILED  - 'fail' response
#define PROTO_INVALID	0x13	// INSYNC/INVALID - 'invalid' response for bad commands
```
We would need to add 
```#define PROTO_PROHIBITED_BY_SILICON_REV		0x14	 // ```
 
* The boot script should clearly indicate the reason for the failure and explain the fundamental incompatibility.

I can refer to https://pixhawk.org/help/errata or incoprate the text. Which would you prefer?

One more thing: The FMUV2 I have is Rev A Silicon - so I will need a new one.

LorenzMeier, 2015-12-01:
Yes, need a new command PROTO_GET_SILICON_VERSION. It should return the binary code and then decode to the ASCII value in Python. We should add a new error code and then show the link to the errata.

The Pixracer you have is silicon rev 3, so you should be good testing with that one.

davids5, 2015-12-04:
@LorenzMeier 

Here is how this is shaping up.

FMUV2 on rev A Silicon
```
Found board 9,0 bootloader rev 5 on /dev/ttyACM0
50583400 00ac2600 00100000 00ffffff ffffffff ffffffff ffffffff ffffffff 543eaedf d70e7861 edb5ca1b 4ad54cd3 61f6bec0 91a83080 a36eab1a d801cc6d 33c585a2 5ef27612 4c41f062 d1f929c8 dd48e75f 7a10adb4 1d873346 a01e4e24 37828b2e 0ee71693 05a1a4ee 1243fc31 5c5b1df9 e5a87db7 d43555ab ee5033e8 e5b35b9f 753005f8 21f74d29 0fd1dba9 e35b042f 85654551 2dcd9604 e01f9a7d ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: PX4
idtype: =00
vid: 000026ac
pid: 00000010
coa: VD6u39cOeGHttcobStVM02H2vsCRqDCAo26rGtgBzG0zxYWiXvJ2EkxB8GLR+SnI3UjnX3oQrbQdhzNGoB5OJDeCiy4O5xaTBaGk7hJD/DFcWx355ah9t9Q1VavuUDPo5bNbn3UwBfgh900pD9HbqeNbBC+FZUVRLc2WBOAfmn0=

sn: 0026001a3432470831323533
chip: 10006419
family: STM32F42x
revision: A
flash 1032192
Erase  : [====================] 100.0%
...
````
FMUV4 Rev A Silicon
```
oaded firmware for b,0, size: 949924 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board b,0 bootloader rev 5 on /dev/ttyACM0
50583400 00ac2600 00100000 00ffffff ffffffff ffffffff ffffffff ffffffff 543eaedf d70e7861 edb5ca1b 4ad54cd3 61f6bec0 91a83080 a36eab1a d801cc6d 33c585a2 5ef27612 4c41f062 d1f929c8 dd48e75f 7a10adb4 1d873346 a01e4e24 37828b2e 0ee71693 05a1a4ee 1243fc31 5c5b1df9 e5a87db7 d43555ab ee5033e8 e5b35b9f 753005f8 21f74d29 0fd1dba9 e35b042f 85654551 2dcd9604 e01f9a7d ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: PX4
idtype: =00
vid: 000026ac
pid: 00000010
coa: VD6u39cOeGHttcobStVM02H2vsCRqDCAo26rGtgBzG0zxYWiXvJ2EkxB8GLR+SnI3UjnX3oQrbQdhzNGoB5OJDeCiy4O5xaTBaGk7hJD/DFcWx355ah9t9Q1VavuUDPo5bNbn3UwBfgh900pD9HbqeNbBC+FZUVRLc2WBOAfmn0=

sn: 0026001a3432470831323533
chip: 10006419
family: STM32F42x
revision: A
flash 1032192

Erase  : [                    ] 0.0%
ERROR: Programing not supported for this verion of Silicon
```

FMUV4 on Rev 3 Silicon

```
Loaded firmware for b,0, size: 949924 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Found board b,0 bootloader rev 5 on /dev/ttyACM0
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 002a003b3135511035333436
chip: 20016419
family: STM32F42x
revision: 3
flash 2080768

Erase  : [====================] 100.0%
Program: [=                   ] 6.8%^C
````

davids5, 2015-12-04:
@LorenzMeier 

Do you want the F1's to report the family and revision?

Options are:
 No
 Generic - Say STM32F1xxx and ? for Revision
 Fully decode

Let me know and I will test pio etc and make sure we are all good - then I will do PR.

ksschwabe, 2015-12-04:
@davids5 : I think to be consistent it would be a good idea to have the F1 chips also report this info. The BL size for the F1s if getting quite large, though, so hopefully it all still fits in the 4kB.

ksschwabe, 2015-12-04:
@davids5 : One more thing... what happens if the silicon revision is increased again? Namely the chip is revised and there is a new silicon revision from 3 to 4. What would the revision be reported as?

Obviously the bootloader would have to updated then for the newer silicon revision, but you would need to know somehow that there is a new silicon revision.

davids5, 2015-12-04:
@ksschwabe 

it will be ? for unknown revs 

Keep in mind the GET_CHIP will always return the binary - so we have a way to see it has changed. But the goal was to encapsulate the decoding to the device that understands it.

For the F1 we are short on time, once my PR is, later today, if you would like to extend the F1 (it will report STM32F1xxx,? ) to fully decode it. 

A PR would be most welcome!

ksschwabe, 2015-12-04:
@davids5: Ok. I'll have a brief look at it this weekend.

davids5, 2015-12-04:
@LorenzMeier 

See https://github.com/PX4/Bootloader/pull/47 and https://github.com/PX4/Firmware/pull/3326

The latter was a fresh rebase on master.

LorenzMeier, 2016-04-23:
Done.

closed 2016-04-23 by LorenzMeier