PX4/PX4-Bootloader #46: BOARD_FORCE_BL on FMUv4 breaks ESC calibration

tridge, 2015-11-26:
The BOARD_FORCE_BL_PORT and PIN is PORTE GPIO14. This is GPIO_GPIO0_INPUT in board_config.h, so it maps to PWM1. That means we get a bunch of pulsed on PWM1 in the bootloader.
These pulses cause ESCs to initialise, which means you can't do proper ESC calibration.
We need to move this functionality to another pin for FMUv4. The only real question is which pin.

ksschwabe, 2015-11-26:
@tridge @LorenzMeier 

What you could do is change the FORCE_BL scheme from using BOARD_FORCE_BL_PIN_OUT and BOARD_FORCE_BL_PIN_IN, to just using BOARD_FORCE_BL_PIN. With the previous scheme, the two pins had to be connected together to force the bootloader. With the latter scheme, the single force BL pin just needs to be held either high or low.

This should solve your problem, since it won't pulse any output through the pin, just read it as an input.

Here is an example from the FMU-v1.
```
# define BOARD_FORCE_BL_PIN             GPIO10
# define BOARD_FORCE_BL_PORT            GPIOA
# define BOARD_FORCE_BL_CLOCK_REGISTER  RCC_AHB1ENR
# define BOARD_FORCE_BL_CLOCK_BIT       RCC_AHB1ENR_IOPAEN
# define BOARD_FORCE_BL_PULL            GPIO_PUPD_PULLUP
# define BOARD_FORCE_BL_STATE           0
```

LorenzMeier, 2015-11-26:
Fixed on master. @ksschwabe it turned out to be a fringe use case and it can prevent the boot of the board depending on what is attached to it. So it seems like we risk "unreliability" from a user / adopter perspective without gaining anything. I think if we continue to support this feature it should go to a test pad which is not connected on normal Pixhawk hardware.

closed 2015-11-26 by LorenzMeier