PX4/PX4-Bootloader #48: Wireless bootloading

LorenzMeier, 2016-01-01:
@ksschwabe This is to track wireless boot loading. Are you aware about this board with Wifi?
https://pixhawk.org/modules/pixracer

Its working now really well with UDP / Wifi, much easier than Bluetooth and it has 100m range with just the very basic PCB antenna. Also note the huge cost reduction and state of the art sensors.

It that of interest to you? What about wireless boot loading and the re-try logic?

dogmaphobic, 2016-01-01:
As it is, the WiFi module is working as a MavLink bridge. It would be simple to create a *mode switch*, which would allow raw data bridging if that becomes necessary. One way or the other, there is no guarantee packets aren't dropped so the protocol must be robust enough to deal with data loss.

ksschwabe, 2016-01-02:
@LorenzMeier I'll have a look at that this week.

liangzelang, 2016-07-16:
Look at this repo  [https://github.com/liangzelang/mavesp8266_PX4bootloader_bridge.git](url)
 I use ESP8266 to upload  firmware for PX4  ,and this project is base on [https://github.com/dogmaphobic/mavesp8266.git](url) .

LorenzMeier, 2021-01-01:
Closing as stale.

closed 2021-01-01 by LorenzMeier