PX4/PX4-Bootloader #50: Low Input on SBUS/PPM/PA10 Causes Stuck Bootup on PX4-V1

ethomas997, 2016-02-09:
There is an issue with the PX4MiniAIO (and other PX4-V1 boards) where the bootup can get stuck when a receiver connected to the SBUS/PPM-IN pin (PA10) is not receiving data from a transmitter.  If that PA10 pin is pulled low at startup, the bootloader code goes into a "wait" mode and stays there until the pin goes high.  It seems that only some receivers work this way.  When I tested with an OpenLRSng-DTFUHF receiver, its PPM output was not pulled low when not receiving data.  Others, however, have reported the issue where the PX4MiniAIO board will not boot up when the transmitter is not turned on.

In the bootloader code, this functionality is described in the comments as "forcing the bootloader."  Interestingly, there is a similar functionality for the PX4-V2 (Pixhawk) boards, but it was disabled in response to [PX4-Bootloader issue #46](https://github.com/PX4/Bootloader/issues/46).  It seems reasonable, then, that this functionality should also be disabled in the PX4-V1 code.  This is accomblished by commenting out the "#BOARD_FORCE_BL..." lines in the "hw_config.h" file, as shown [here](https://github.com/ethomas997/PX4_Bootloader/commit/a9fb07a1f3a15c3d9c13806dd403cf7c3c6d9b06):
```
/*
 * Uncommenting this allows to force the bootloader through
 * the PPM-in pin. Some receivers pull their PPM output low
 * when the transmitter is off, resulting in a stuck bootup,
 * so this feature is best disabled.
 *
 * # define BOARD_FORCE_BL_PIN             GPIO10
 * # define BOARD_FORCE_BL_PORT            GPIOA
 * # define BOARD_FORCE_BL_CLOCK_REGISTER  RCC_AHB1ENR
 * # define BOARD_FORCE_BL_CLOCK_BIT       RCC_AHB1ENR_IOPAEN
 * # define BOARD_FORCE_BL_PULL            GPIO_PUPD_PULLUP
 * # define BOARD_FORCE_BL_STATE           0
 */
```
I have a branch with this modification posted [here](https://github.com/ethomas997/PX4_Bootloader/tree/NoV1ForceBlPin).  I can post it as a PR if this change is wanted.

I also have information posted here:  http://www.etheli.com/APM/PX4MiniAIO

--ET

mhkabir, 2016-02-10:
This indeed fixes the RC-in related issues. Would you mind sending in a PR? I believe that a code review would be much simpler in that case. 

Thanks!

ethomas997, 2016-02-10:
Sure, PR is here:  https://github.com/PX4/Bootloader/pull/51

--ET

closed 2016-02-10 by LorenzMeier