PX4/PX4-Bootloader #53: Disable the AHB peripheral clocks

liangzelang, 2016-04-23:
there is a sentence :
RCC_AHB1ENR = 0x00100000; // XXX Magic reset number from STM32F4x reference manual
in the board_init() In the main_f4.c. 
 However ,i found it is wrong after look up the stm32F4xx refence manual.

LorenzMeier, 2016-04-23:
Could you please instead tell us what you think the correct value is? Why did you look it up in the first place? What problem are you trying to fix?

LorenzMeier, 2016-05-05:
Can you have a look?

liangzelang, 2016-05-10:
sorry to forget check your message.   In fact , I don't know what the coder want to do. 
1.    the 20 bit of AHB1ENR  is about CCM data RAM clock.   if you want to disable it  
        RCC_AHB1ENR = 0<<20;
2.  according to the annotate， the coder want to disable the AHB clock.  but  i don't know how to close the clock.

davids5, 2016-06-08:
[image]

closed 2016-06-08 by davids5

liangzelang, 2016-06-11:
Got it ， thank U