PX4/PX4-Bootloader #59: Button management on TAP

LorenzMeier, 2016-08-10:
@davids5 I can't seem to be able to power on via the button only and it feels generally confusing from an user perspective.

LorenzMeier, 2016-08-10:
Feels like a potential debouncing issue. Same within the FW.

LorenzMeier, 2016-08-10:
It feels like the power should be actively shut off here - but this seems just to wait.

```C
/* Here we check for the app setting the POWER_DOWN_RTC_SIGNATURE
	 * in this case, we reset the signature and wait to die
	 */
	if (board_get_rtc_signature() == POWER_DOWN_RTC_SIGNATURE) {
		board_set_rtc_signature(0);

		while (1);
	}
```

davids5, 2016-08-10:
For me On was never a problem. Off was - On my HW the following works as expected. I wonder if HW revision may be the cause of what you are seeing?

The current design and code does the following:

OFF ---- Any Button Press or USB -----> ON [if FW Valid] Boot 

Note: The BL has no button detection and turn off logic.

ON --------- A Button Press > MS_PWR_BUTTON_DOWN  ---> Begin Shut down*

*Once booted to FW - A button press lasting >= MS_PWR_BUTTON_DOWN  (.75 Sec)  will begin to shut off the power. It will indicate so via the top LED for .75.  The following overly complicated control is because the HW has a diode in the circuit, and hence no active way to discharge the node that enables power. So the FW sets the signature and does a HW reset, this changes the power signal to an input. (Which is the best we can do because of the diode). The bootloader is activated by the HW reset, and because of the signature, (which it clears) it waits to brown out.  Without setting the Power signal active.

We should discuss what you are seeing.

davids5, 2016-08-11:
@LorenzMeier - All that is left to close this is determining  if MS_PWR_BUTTON_DOWN  is adequate to avoid  accidental activation.

LorenzMeier, 2021-01-01:
Closing as stale.

closed 2021-01-01 by LorenzMeier