PX4/PX4-Bootloader #82: Board always stays in bootloader even when USB not connected

ksschwabe, 2017-04-27:
Since updating to the latest bootloader, I have the problem that the board always waits in the bootloader until the timeout has elapsed, even if the USB cable is not connected - the board is powered separately.

It seems that this behaviour was introduced in commit 1770cfd162e5e8e895cc6f35ecf8c48543fffed1.
I understand that we should not be pulling the Vbus sense line low using the internal pull downs. But leaving it floating seems to result in the check for whether the Vbus is connected to be always true.
The check `if (gpio_get(GPIOA, GPIO9) != 0)` in main_f4.c always returns true.

I think if we are supposed to be using the hardware detection of the VBUS sensing, then we should probably be setting the VBUSBSEN bit in the OTG_FS_GCCFG register. But I haven't worked out how we are supposed to then use this to detect the USB.
```
OTG_FS_GCCFG |= OTG_GCCFG_VBUSBSEN;
``` 

The other option is to always disable the hardware VBUS sensing (always set the NOVBUSSENS bit) and then just read the PA9 pin as usual.

LorenzMeier, 2017-12-31:
Did you progress on that?

LorenzMeier, 2018-01-04:
@ksschwabe Could you comment on this?

ksschwabe, 2018-01-04:
@LorenzMeier: Sorry for the delay in replying (holiday). I believe this is still a problem. I will confirm tomorrow and get back to you.

ksschwabe, 2018-01-08:
@LorenzMeier: This still seems to be a problem for me on my custom board, but not on a Px4FMU-v3. Up until now, I had just reverted commit 1770cfd, and "ignored" the problem on my custom board

My "custom" board also uses an STM32F427 and the bootloader logic should work exactly the same as for the Px4FMU-v2 and -v3 boards, so I am not sure where my problem is coming from.

ksschwabe, 2018-01-19:
@LorenzMeier @davids5 : I have finally taken the time to see what the problem is.

In commit 1770cfd, we said that the VBUS sense pin, PA9, should be left "floating" when configured configured for VBUS sensing, as specified in the Datasheet.
[image]

However, when we perform the check for whether the USB is plugged in (PA9 is high), the USB has not yet been configured. This means that PA9 is still acting as a normal floating input. This means that 
when the USB is not plugged in, PA9 could be read as HIGH.
Lines 781 - 790 in main_f4.c
```
#if defined(BOARD_USB_VBUS_SENSE_DISABLED)
	try_boot = false;
#else
	if (gpio_get(GPIOA, GPIO9) != 0) {

		/* don't try booting before we set up the bootloader */
		try_boot = false;
	}
#endif
#endif
```

The USB interface is only configured later in lines 831-833, AFTER we have checked PA9.
```
#if INTERFACE_USB
	cinit(BOARD_INTERFACE_CONFIG_USB, USB);
#endif
```

@davids5: I think we can fix this in one of two ways, either we 
  * configure the interfaces at the beginning with the board_init(), or just after it; or
  * configure PA9 as an input with a pulldown in the beginning, and perform your check as to whether to stay in the bootloader (check if PA9 is high), and then just before configuring the USB interface, reset PA9 to having a floating input.

What do you think?

davids5, 2018-01-19:
@ksschwabe -  I think it might be best to implement a function say "bl_is_usb_actitive()" in the f1,f4,f7 c files and init, check, deinit on the SoC's that need it to be that way.

ksschwabe, 2018-01-19:
@davids5 : How about configuring PA9 to be a pulled down input in board_init(), like it used to be, and then in `usb_cinit(void)` configure it to be a floating input as required when initialising/configuring the USB. This means that whenever usb_cinit is called, it ensures that PA9 is configured correctly for USB.

davids5, 2018-01-19:
I would prefer the abstraction via the API but your proposal is fine.

davids5, 2018-01-19:
The USB check if we must stay in bootloader becomes a function (bool bl_is_USB_active()) it does the 

1. init or the port
2. Read of port
3. deinit of port
4 returns the read value.

all in one place.

Then for the F1..F7 implement the code per the data sheet requirements and just call it with the online if def is.

It would just help make it clear what we are doing as apposed to distributed pieces of code in many places.

ksschwabe, 2018-01-19:
Ok, I see. Then the config of PA9 in board_init() will be removed, and the usb_cinit() and uart_cinit() remain where they currently are?

I'll work on that and prepare PR for you on Monday.

ksschwabe, 2018-01-23:
PR  #96

LorenzMeier, 2021-01-01:
Closing as stale.

closed 2021-01-01 by LorenzMeier