PX4/PX4-Bootloader #84: bootloader's uart update is not ok

yazhouren2012, 2017-06-07:
hi, when i use usb to update nuttx, it's ok,
but when i use uart to update nuttx, it's not ok, so i debug the bootloader, i find that the function uart_cout() can't output anything, may be there is somewrong with this serial port 2.

yazhouren2012, 2017-06-07:
MY hardware is pixhawk v2

kenshin1988, 2020-09-29:
USART2 upload not working on Pixhawk4(TELEM1) with px_uploader.py.
Works well on USB using px_uploader.py. Anything we are missing? Repeated message showing
"If the board does not respond, unplug and replug the USB connector"

kenshin1988, 2020-09-29:
I compiled and updated the FMU bootloader and tried to program again. I am able to go a step ahead now. Erase is completed with 100%. Program gets stuck at 0% and exits after a while with the message "ERROR: unexpected 18 instead of ISYNC"

LorenzMeier, 2021-01-01:
Closing as stale.

jlecoeur, 2021-01-20:
This is still an issue as of today. I suggest reopening.
@LorenzMeier @davids5 is UART bootlader functional on FMUv5 and FMUv5x ?

davids5, 2021-01-20:
AFAIK yes. 

V5 it is on USART2_TELEM1 115200 BPS (not tested recently) Reopening for testing.
V5X  UART5_TELEM2            1500000BPS (Tested recently)

jlecoeur, 2021-02-08:
solved in https://github.com/PX4/PX4-Autopilot/pull/16766

closed 2021-02-08 by jlecoeur