PX4/PX4-Bootloader #87: Does not work on STM32F7

wenfenglin, 2017-09-02:
I built px4fmuv5_bl for my STM32F765VI but it did not works. 

Then I have found that the bootloader uses the library for STM32F4 instead of STM23F7. 

With a few modification (some register addresses and bit masks) I can run the bootloader and flash the firmware via UART, but USB is more complicated than UART and I haven't solved the problem yet.

Have you guys tested on an F7 board yet?

wenfenglin, 2017-09-02:
Never mind. USB works fine after I reconfigured the RCC

closed 2017-09-02 by wenfenglin

ksschwabe, 2017-09-02:
@wenfenglin : Out of interest what change did you have to make to the RCC to get the USB to work on the F7?

wenfenglin, 2017-09-02:
@ksschwabe 
On my board, HSE is generated by a external clock of 8MHz (i.e. bypass mode). In order to provide a 48MHz clock for the USB, I changed the plln (at ine 139 of File main_f7.c) from 216 to 432 and it works.

BTW, it seems that UART library of F4 is not compatible with F7 (The register map of F7 looks pretty much the same as F3 instead of F3).