PX4/PX4-Bootloader #88: RCC Clock setup on F7 compilation not using OSC_FREQ

ksschwabe, 2017-09-04:
@davids5: I noticed that the clock setup for the FMUV5_bl is not using the OSC_FREQ define in the clock config.

```
/* standard clocking for all F7 boards */
static const struct rcc_clock_scale clock_setup = {
	.pllm = 8,
	.plln = 216,
	.pllp = 2,
	.pllq = 9,
	.pllr = 2,
	.hpre = RCC_CFGR_HPRE_DIV_NONE,
	.ppre1 = RCC_CFGR_PPRE_DIV_4,
	.ppre2 = RCC_CFGR_PPRE_DIV_2,
	.power_save = 0,
	.flash_config = FLASH_ACR_ICE | FLASH_ACR_DCE | FLASH_ACR_LATENCY_5WS,
	.apb1_frequency = 54000000,
	.apb2_frequency = 108000000,
};
```

Going through the calculation it looks like the OSC_FREQ is indeed 16MHZ (as defined in hw_config.h) - this would mean that pllm should be OSC_FREQ and plln should be 432. Is this correct?

I am surprised that the OSC_FREQ is 16MHz. Is it not using the standard 24MHz crystal that we use on the other boards.

davids5, 2017-09-04:
I believe if you do a careful comparison between the PLL that used on the F7 and the PLL that used on the F4 you will see that there is an extra / 2 in the vco of F7.

davids5, 2017-09-04:
it's a holiday here. When I can get back in front of a computer. I will post better response

davids5, 2017-09-05:
@ksschwabe 

Ok now that I am looking at it and realized I was thinking of the Kinetis K66 vs K64 PLL when I made the incorrect comment about the divide by 2.

The original F7 bring up was done on the Nucleo144 that had an 8Mhz xtal. From an EMI standpoint I would like to keep the value as low as possible. But the first proto that was done had a 16Mhz xtal by mistake on it. I adjusted the FMUv5 design to accommodate that board. I think the 24Mhz and 16Mhz pricing is equivalent. Was there some other concern?

The values were chosen to match the one used in the board config below for the PX4 Firmware.
The solution is the same with a lower multiply. 
`PLL_VCO = (16,000,000 / 8) * 216 = 432 MHz  ==  (16,000,000 / 16) * 432 = 432 MHz`

But you are correct it is not driven by the OSC_FREQ. 

 ```
 *  PLL source is HSE = 16,000,000
 *
 * PLL_VCO = (STM32_HSE_FREQUENCY / PLLM) * PLLN
 * Subject to:
 *
 *     2 <= PLLM <= 63
 *   192 <= PLLN <= 432
 *   192 MHz <= PLL_VCO <= 432MHz
 *
 * SYSCLK  = PLL_VCO / PLLP
 * Subject to
 *
 *   PLLP = {2, 4, 6, 8}
 *   SYSCLK <= 216 MHz
 *
 * USB OTG FS, SDMMC and RNG Clock = PLL_VCO / PLLQ
 * Subject to
 *   The USB OTG FS requires a 48 MHz clock to work correctly. The SDMMC
 *   and the random number generator need a frequency lower than or equal
 *   to 48 MHz to work correctly.
 *
 * 2 <= PLLQ <= 15
 */

/* Highest SYSCLK with USB OTG FS clock = 48 MHz
 *
 * PLL_VCO = (16,000,000 / 8) * 216 = 432 MHz
 * SYSCLK  = 432 MHz / 2 = 216 MHz
 * USB OTG FS, SDMMC and RNG Clock = 432 MHz / 9 = 48 MHz
 */

#define STM32_PLLCFG_PLLM       RCC_PLLCFG_PLLM(8)
#define STM32_PLLCFG_PLLN       RCC_PLLCFG_PLLN(216)
#define STM32_PLLCFG_PLLP       RCC_PLLCFG_PLLP_2
#define STM32_PLLCFG_PLLQ       RCC_PLLCFG_PLLQ(9)
```

ksschwabe, 2017-09-05:
My concern was just related to the fact that OSC_FREQ was not being used in the clock_setup in the bootloader, so it broke with the convention used for the F1, F3 and F4 boards. That's all.

(I was just surprised that the OSC_FREQ was 16MHz instead of the usual 24MHz that had been used up until now.)

LorenzMeier, 2021-01-01:
Closing as stale.

closed 2021-01-01 by LorenzMeier