PX4/PX4-Bootloader #93: mRo (AUAV) X-2.1 stuck in Bootloader

tops4u, 2017-12-29:
**BUG**

I have an mRo X-2.1 which is connected on TELEM 1 (UART-2) to a 433MHz Radio. When the Flight Controller is only USB Powered, the Radio does not get any Power, so the RX Line of UART-2 will be LOW. This causes the X-2.1 to stay in Bootloader Mode. As soon as the Connection to the Radio is removed or a Battery is connected - even only for less than a second - the Device will Boot. I have seen that the configuration in hwconfig.h is identical to FMUv2 Devices but I tried to Short UART 2 (Serial 1) RX - GND on a AUAV-X2 as well as on a Dropix v2.1 Controller and they do both boot.

tops4u, 2017-12-31:
Confirmed, that this is an issue with the Serial Port. Setting "SERIAL_BREAK_DETECT_DISABLED" for AUAV X-2.1 did the job, but is only a Workaround.

LorenzMeier, 2017-12-31:
@tops4u Could you please also cross test and revert this to see if it is related to generally floating voltages?
https://github.com/PX4/Bootloader/commit/1770cfd162e5e8e895cc6f35ecf8c48543fffed1

tops4u, 2017-12-31:
@LorenzMeier I did not perform this Crosstest regarding Voltages, but doubt this to be a Problem. Checking the SERIAL_BREAK_DETECT_DISABLED Flag which solved the Problem for mRo, it is only used for the Serial Line Break Logic. As my Testing abilities are limited, I do not want to mess up things ;-)

tops4u, 2018-01-01:
Fixed with PR #94.

closed 2018-01-01 by tops4u