PX4/PX4-Bootloader #111: Make command doesn't run

ThomasH3, 2018-08-10:
I'm getting the following error message while running the make command.
make (e=2): "Can't find the file".
make[1]: *** [build/nxphlitev3_bl/bl.o] Error 2
make: *** [nxphlitev3_bl] Error 2

prtkj, 2018-08-21:
I'm too getting error on running the make command for the booloader.
Though the px4 firmware builds successfully.
What could I be missing ?
ERROR 
make --no-print-directory -f  Makefile.k66 TARGET_HW=NXPHLITE_V3  LINKER_FILE=kinetisk66.ld TARGET_FILE_NAME=nxphlitev3_bl
Generating object build/nxphlitev3_bl/kinetis/cdcacm.o
arm-none-eabi-gcc -c -MMD -std=gnu99 -Os -g -Wundef -Wall -fno-builtin -I./libopencm3/include -I./. -ffunction-sections -nostartfiles -lnosys -Wl,-gc-sections -Wl,-g -Werror -DSTM32F4 -DTOWER=0 -DFREEDOM=0  -g -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -specs=nano.specs -specs=nosys.specs -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/platform/devices/MK66F18 -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/targets/MK66F18/src -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/autobaud -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/usb/include -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/usb/osa -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/usb/device -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/usb/device/class -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/bm_usb -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/include -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/platform -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/platform/devices -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/platform/CMSIS/Include -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/drivers -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/startup -I lib/kinetis/NXP_Kinetis_Bootloader_2_0_0/src/drivers/common -DTARGET_HW_NXPHLITE_V3 -DBL_CONFIG_USB_CDC=1 -DBL_CONFIG_HS_USB_HID=0 -DBL_CONFIG_HS_USB_MSC=0 -DBL_TARGET_FLASH -DDEBUG=0 -DNDEBUG=1 -DUSB_STACK_BM -DCPU_MK66FN2M0VMD18=1 -DCPU_IS_ARM_CORTEX_M4=1 -Tkinetisk66.ld  -o build/nxphlitev3_bl/kinetis/cdcacm.o kinetis/cdcacm.c
kinetis/cdcacm.c:38:10: fatal error: fsl_device_registers.h: No such file or directory
 #include "fsl_device_registers.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
rules.mk:28: recipe for target 'build/nxphlitev3_bl/kinetis/cdcacm.o' failed
make[1]: *** [build/nxphlitev3_bl/kinetis/cdcacm.o] Error 1
Makefile:77: recipe for target 'nxphlitev3_bl' failed
make: *** [nxphlitev3_bl] Error 2

ksschwabe, 2018-08-21:
@davids5: It looks as if the fsl_device_registers.h file required by kinetis/cdcacm.c (original commit adding kinetis/cdcacm.c 000ba88f9db897f7f270a3a78c8c06e290927475) was never committed into the repo.

@ThomasH3 @prtkj : You can still build your required bootloader (as long as it isn't one for one of the kinetis targets) by running `make <target name>`, e.g. `make px4fmuv3_bl`.

prtkj, 2018-08-21:
@ksschwabe : Actually I tried that as well. I could successfully build the **px4fmuv5_bl** . It doesn't show any error. But the **px4fmuv2_bl** doesn't build successfully 
It shows the following error - 
Generating object build/px4fmuv2_bl/main_f4.o
arm-none-eabi-gcc -c -MMD -std=gnu99 -Os -g -Wundef -Wall -fno-builtin -I./libopencm3/include -I./. -ffunction-sections -nostartfiles -lnosys -Wl,-gc-sections -Wl,-g -Werror -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DTARGET_HW_PX4_FMU_V2 -DSTM32F4 -Tstm32f4.ld -Llibopencm3/lib -lopencm3_stm32f4  -o build/px4fmuv2_bl/main_f4.o main_f4.c
main_f4.c:179:3: error: 'const struct rcc_clock_scale' has no member named 'power_save'
  .power_save = 0,
   ^~~~~~~~~~
rules.mk:28: recipe for target 'build/px4fmuv2_bl/main_f4.o' failed
make[1]: *** [build/px4fmuv2_bl/main_f4.o] Error 1
Makefile:86: recipe for target 'px4fmuv2_bl' failed
make: *** [px4fmuv2_bl] Error 2

prtkj, 2018-08-21:
I wanted to build bootloader for STM32 F4 series.That is for Pixhawk 1.

ksschwabe, 2018-08-21:
@prtkj : The **px4fmuv2_bl** compiles on my system. Have you updated the libopencm3 submodule?

To update the submodule, run the following commands
```
git submodule init
git submodule update
make clean
make px4fmuv2_bl
```

prtkj, 2018-08-21:
@ksschwabe : I tried the commands but still couldn't build.
But then I cloned the Bootloader repository again, ran these commands and could successfully build it !!!
Thanks a lot !!

ThomasH3, 2018-08-21:
It doens't work.

C:\Bootloader>make clean
cd libopencm3 && make --no-print-directory clean && cd ..
make[1]: Entering directory `c:/Bootloader/libopencm3'
  GENHDR  include/libopencm3/stm32/f0/irq.json
/bin/env: python: No such file or directory
make[1]: *** [include/libopencm3/stm32/f0/irq.json.genhdr] Error 127
make[1]: Leaving directory `c:/Bootloader/libopencm3'
make: *** [clean] Error 2

C:\Bootloader>make px4fmuv2_bl
(./Tools/check_submodules.sh)
/bin/env: bash: No such file or directory
make: *** [checksubmodules] Error 127

davids5, 2018-08-21:
@ThomasH3 What OS are you building on? It looks like Windows. I have never even tried to build this on Windows but that looks to be your issue.  Have you tried it on linux?

ThomasH3, 2018-08-22:
Error message in Linux:
make[2]: arm-none-eabi-gcc: Command not found
../Makefile.include:40: recipe for target 'ccm.o' failed
make[2]: *** [ccm.o] Error 127
/bin/sh: 1: [: .stamp_failure_lib_efm32_ezr32wg: unexpected operator
make[1]: Leaving directory '/home/tom/Downloads/bootloader/libopencm3'
make --no-print-directory -f  Makefile.f4 TARGET_HW=PX4_FMU_V2  LINKER_FILE=stm32f4.ld TARGET_FILE_NAME=px4fmuv2_bl
mkdir -p build/px4fmuv2_bl build/px4fmuv2_bl/./ build/px4fmuv2_bl/stm32/
Generating object build/px4fmuv2_bl/bl.o
arm-none-eabi-gcc -c -MMD -std=gnu99 -Os -g -Wundef -Wall -fno-builtin -I./libopencm3/include -I./. -ffunction-sections -nostartfiles -lnosys -Wl,-gc-sections -Wl,-g -Werror -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DTARGET_HW_PX4_FMU_V2 -DSTM32F4 -Tstm32f4.ld -Llibopencm3/lib -lopencm3_stm32f4  -o build/px4fmuv2_bl/bl.o bl.c
make[1]: arm-none-eabi-gcc: Command not found
rules.mk:28: recipe for target 'build/px4fmuv2_bl/bl.o' failed
make[1]: *** [build/px4fmuv2_bl/bl.o] Error 127
Makefile:86: recipe for target 'px4fmuv2_bl' failed
make: *** [px4fmuv2_bl] Error 2

davids5, 2018-08-22:
@ThomasH3 

**make[2]: arm-none-eabi-gcc: Command not found** You do not have a toolchain installed.

Please follow the dev guide @ https://dev.px4.io/en/setup/dev_env.html for linux.

prtkj, 2018-08-22:
@ThomasH3  as david says, try setting up the toolchain as mentioned in the link for linux. Follow the development of Pixhawk/NuttX and try make the px4fmuv2_bl .

ThomasH3, 2018-08-22:
Got it.

Thank You !

closed 2018-08-22 by ThomasH3