PX4/PX4-Bootloader #120: Robustify bootloader protocol

RomanBapst, 2018-11-13:
**Problem:**
Currently, when data from the px_uploader to the bootloader is not transmitted correctly the crc check in the bootloader fails and the firmware update therefore fails as well. This basically never happens when using the USB interface but it happens frequently when using the UART interface with relatively high baudrates.
When using a baudrate of 921600 we frequently see this problem and updating firmware via this interface is therefore very unreliable.

We were thinking about making the protocol between the bootloader and the px_uploader more robust so that we don't need to rely on 100% correct data transmission in order to update the firmware. Mostly it's just a few bytes which are not transmitted correctly but that is enough to make the update fail in the current state. We were thinking of adding at least a couple of retries.

@davids5 What do you think about this?

ksschwabe, 2018-11-16:
@RomanBapst : This is related to Issue #48 and PR #61 (wireless bootloading/handling bit and byte drops, etc.). I think PR #61 was already quite far along, but still needed some polishing and code review.

LorenzMeier, 2021-01-01:
This has been addressed since.

closed 2021-01-01 by LorenzMeier