PX4/PX4-Bootloader #160: Error make Bootloader (Toolchain Windows 10)

astekachev, 2020-01-09:
Hi!
When I try to build Bootloader through the PX4 Toolchain on the command line through the 'make' or "make px4fmuv5_bl" commands, the compiler throws an error "cannot find -lopencm3_stm32f4"

I put the log below

$ make px4fmuv5_bl
(./Tools/check_submodules.sh)
Checked libopencm3 submodule, correct version found
Checked lib/kinetis/NXP_Kinetis_Bootloader_2_0_0 submodule, correct version found
make -C libopencm3 lib
make[1]: вход в каталог «/cygdrive/c/PX4/home/Bootloader/libopencm3»
  GENHDR  include/libopencm3/stm32/f0/irq.json
  GENHDR  include/libopencm3/stm32/f1/irq.json
  GENHDR  include/libopencm3/stm32/f2/irq.json
  GENHDR  include/libopencm3/stm32/f3/irq.json
  GENHDR  include/libopencm3/stm32/f4/irq.json
  GENHDR  include/libopencm3/stm32/f7/irq.json
  GENHDR  include/libopencm3/stm32/l0/irq.json
  GENHDR  include/libopencm3/stm32/l1/irq.json
  GENHDR  include/libopencm3/stm32/l4/irq.json
  GENHDR  include/libopencm3/lpc13xx/irq.json
  GENHDR  include/libopencm3/lpc17xx/irq.json
  GENHDR  include/libopencm3/lpc43xx/m4/irq.json
  GENHDR  include/libopencm3/lpc43xx/m0/irq.json
  GENHDR  include/libopencm3/lm3s/irq.json
  GENHDR  include/libopencm3/efm32/tg/irq.json
  GENHDR  include/libopencm3/efm32/g/irq.json
  GENHDR  include/libopencm3/efm32/lg/irq.json
  GENHDR  include/libopencm3/efm32/gg/irq.json
  GENHDR  include/libopencm3/efm32/hg/irq.json
  GENHDR  include/libopencm3/efm32/wg/irq.json
  GENHDR  include/libopencm3/efm32/ezr32wg/irq.json
  GENHDR  include/libopencm3/sam/3a/irq.json
  GENHDR  include/libopencm3/sam/3n/irq.json
  GENHDR  include/libopencm3/sam/3s/irq.json
  GENHDR  include/libopencm3/sam/3u/irq.json
  GENHDR  include/libopencm3/sam/3x/irq.json
  GENHDR  include/libopencm3/sam/4l/irq.json
  GENHDR  include/libopencm3/sam/d/irq.json
  GENHDR  include/libopencm3/vf6xx/irq.json
  BUILD   lib/stm32/f0
  AR      libopencm3_stm32f0.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f0.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f0.a] Error 1
  BUILD   lib/stm32/f1
  AR      libopencm3_stm32f1.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f1.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f1.a] Error 1
  BUILD   lib/stm32/f2
  AR      libopencm3_stm32f2.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f2.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f2.a] Error 1
  BUILD   lib/stm32/f3
  AR      libopencm3_stm32f3.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f3.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f3.a] Error 1
  BUILD   lib/stm32/f4
  AR      libopencm3_stm32f4.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f4.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f4.a] Error 1
  BUILD   lib/stm32/f7
  AR      libopencm3_stm32f7.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f7.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32f7.a] Error 1
  BUILD   lib/stm32/l0
  AR      libopencm3_stm32l0.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32l0.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_stm32l0.a] Error 1
  BUILD   lib/stm32/l1
  AR      libopencm3_stm32l1.a
[... 74 lines cut ...]
  BUILD   lib/sam/3x
  AR      libopencm3_sam3x.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_sam3x.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_sam3x.a] Ошибка 1
  BUILD   lib/sam/4l
  AR      libopencm3_sam4l.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_sam4l.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_sam4l.a] Error 1
  BUILD   lib/sam/d
  AR      libopencm3_samd.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_samd.a: No such file or directory
make[2]: *** [../../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_samd.a] Error 1
  BUILD   lib/vf6xx
  AR      libopencm3_vf6xx.a
C:\PX4\toolchain\gcc-arm\bin\arm-none-eabi-ar.exe: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_vf6xx.a: No such file or directory
make[2]: *** [../Makefile.include:37: /cygdrive/c/PX4/home/Bootloader/libopencm3/lib/libopencm3_vf6xx.a] Error 1
/bin/sh: line 0: [: too many arguments
make [1]: exit the directory «/cygdrive/c/PX4/home/Bootloader/libopencm3»
make --no-print-directory -f  Makefile.f7 TARGET_HW=PX4_FMU_V5 LINKER_FILE=stm32f7.ld TARGET_FILE_NAME=px4fmuv5_bl
arm-none-eabi-gcc -o build/px4fmuv5_bl/px4fmuv5_bl.elf build/px4fmuv5_bl/bl.o build/px4fmuv5_bl/stm32/cdcacm.o build/px4fmuv5_bl/stm32/usart.o build/px4fmuv5_bl/main_f7.o  -std=gnu99 -Os -g -Wundef -Wall -fno-builtin -I./libopencm3/include -I./. -ffunction-sections -nostartfiles -lnosys -Wl,-gc-sections -Wl,-g -Werror -g -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -DTARGET_HW_PX4_FMU_V5 -Tstm32f7.ld -Llibopencm3/lib -lopencm3_stm32f7 -lopencm3_stm32f4 -DSTM32F7
c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lopencm3_stm32f7
c:/px4/toolchain/gcc-arm/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lopencm3_stm32f4
collect2.exe: error: ld returned 1 exit status
make[1]: *** [rules.mk:41: build/px4fmuv5_bl/px4fmuv5_bl.elf] Error 1
make: *** [Makefile:116: px4fmuv5_bl] Error 2

astekachev, 2020-01-10:
Я решил эту проблему путем редактирования файлов Make. Позже отпишу как решать, если лень не будет может еще и на английский переведу. Интересно как он вообще компилируется, если в рецептах куча ошибок в путях.

astekachev, 2020-01-12:
I solved this problem by editing the Make files. I wonder how Bootloader generally compiles if there are a lot of errors in Make files.
1. In the file ... \ Bootloader \ libopencm3 \ lib \ Make.include, you need to fix the location of the library files and their extension.
$ (SRCLIBDIR) / $ (LIBNAME) .a: $ (OBJS)
@printf "AR $ (LIBNAME) .a \ n"
# $ (Q) $ (AR) $ (ARFLAGS) "$ @" $ (OBJS)
$ (Q) $ (AR) $ (ARFLAGS) $ (LIBNAME) .a $ (OBJS)

After attempting to compile, library files of the * .a type are compiled, but errors are generated in their location. The following must be done.

2. In the file ... \ Bootloader \ Make.F1
FLAGS + = -mthumb -mcpu = cortex-m3 \
       -DTARGET_HW _ $ (TARGET_HW) \
       -DSTM32F1 \
-T $ (LINKER_FILE) \
-L $ (LIBOPENCM3) / lib / stm32 / F1 \
-lopencm3_stm32f1 \
$ (EXTRAFLAGS)
3. In the file ... \ Bootloader \ Make.F3
FLAGS + = -mthumb -mcpu = cortex-m4 -mfloat-abi = hard -mfpu = fpv4-sp-d16 \
       -DTARGET_HW _ $ (TARGET_HW) \
       -DSTM32F3 \
       -T $ (LINKER_FILE) \
-L $ (LIBOPENCM3) / lib / stm32 / F3 \
-lopencm3_stm32f3 \
        $ (EXTRAFLAGS)
4. In the file ... \ Bootloader \ Make.F4
FLAGS + = -mthumb -mcpu = cortex-m4 -mfloat-abi = hard -mfpu = fpv4-sp-d16 \
       -DTARGET_HW _ $ (TARGET_HW) \
       -DSTM32F4 \
       -T $ (LINKER_FILE) \
-L $ (LIBOPENCM3) / lib / stm32 / F4 \
-lopencm3_stm32f4 \
        $ (EXTRAFLAGS)
5. In the file ... \ Bootloader \ Make.F7
FLAGS + = -g -mthumb -mcpu = cortex-m7 -mfloat-abi = hard -mfpu = fpv5-sp-d16 \
-DTARGET_HW _ $ (TARGET_HW) \
-DSTM32F7 \
-T $ (LINKER_FILE) \
-L $ (LIBOPENCM3) / lib / stm32 / F7 \
-lopencm3_stm32f7 \
-T $ (LINKER_FILE) \
-L $ (LIBOPENCM3) / lib / stm32 / F4 \
-lopencm3_stm32f4 \
$ (EXTRAFLAGS)

After all instructions have been adjusted, Bootloader compiles without errors through the PX4 Toolchaine console

davids5, 2020-02-03:
@astekachev please see https://github.com/PX4/Bootloader/pull/162#issuecomment-581623840 and retest.

bg8wj, 2020-08-05:
> I solved this problem by editing the Make files. I wonder how Bootloader generally compiles if there are a lot of errors in Make files.
> 
> 1. In the file ... \ Bootloader \ libopencm3 \ lib \ Make.include, you need to fix the location of the library files and their extension.
> 
> ```
> $(SRCLIBDIR)/$(LIBNAME).a: $(OBJS)
> 	@printf "  AR      $(LIBNAME).a\n"
> 	#$(Q)$(AR) $(ARFLAGS) "$@" $(OBJS)
> 	$(Q)$(AR) $(ARFLAGS) $(LIBNAME).a $(OBJS)
> ```
> 
> After attempting to compile, library files of the * .a type are compiled, but errors are generated in their location. The following must be done.
> 
> 1. In the file ... \ Bootloader \ Make.F1
> 
> ```
> FLAGS		+= -mthumb -mcpu=cortex-m3\
>        -DTARGET_HW_$(TARGET_HW) \
>        -DSTM32F1 \
> 		   -T$(LINKER_FILE) \
> 				-L$(LIBOPENCM3)/lib/stm32/F1 \
> 				-lopencm3_stm32f1 \
> 		   $(EXTRAFLAGS)
> ```
> 
> 1. In the file ... \ Bootloader \ Make.F3
> 
> ```
> FLAGS 		+= -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 \
>        -DTARGET_HW_$(TARGET_HW) \
>        -DSTM32F3 \
>        -T$(LINKER_FILE) \
> 		   -L$(LIBOPENCM3)/lib/stm32/F3 \
> 		   -lopencm3_stm32f3 \
>         $(EXTRAFLAGS)
> ```
> 
> 1. In the file ... \ Bootloader \ Make.F4
> 
> ```
> FLAGS		+= -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 \
>        -DTARGET_HW_$(TARGET_HW) \
>        -DSTM32F4 \
>        -T$(LINKER_FILE) \
> 		   -L$(LIBOPENCM3)/lib/stm32/F4 \
> 		   -lopencm3_stm32f4 \
>         $(EXTRAFLAGS)
> ```
> 
> 1. In the file ... \ Bootloader \ Make.F7
> 
> ```
> FLAGS			+= -g -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-sp-d16 \
> 				-DTARGET_HW_$(TARGET_HW) \
> 				-DSTM32F7 \
> 				-T$(LINKER_FILE) \
> 					-L$(LIBOPENCM3)/lib/stm32/F7 \
> 					-lopencm3_stm32f7 \
> 				-T$(LINKER_FILE) \
> 					-L$(LIBOPENCM3)/lib/stm32/F4 \
> 					-lopencm3_stm32f4 \
> 				$(EXTRAFLAGS)
> ```
> 
> After all instructions have been adjusted, Bootloader compiles without errors through the PX4 Toolchaine console
follower up but same the build log like this: 
BUILD   lib/stm32/f0
/bin/sh: -c:行1: 寻找匹配的 `"' 是遇到了未预期的文件结束符
/bin/sh: -c:行2: 语法错误: 未预期的文件结尾
make[1]: *** [Makefile:69：lib/stm32/f0] 错误 1
make[1]: 离开目录“/cygdrive/d/Program/px4/home/bootloader/libopencm3”
make: *** [Makefile:202：libopencm3] 错误 2
win10 office Windows Cygwin system，win10 System language build px4 no problem.

LorenzMeier, 2021-01-01:
It might work now - if not, please use the Linux or Mac OS toolchain.

closed 2021-01-01 by LorenzMeier