PX4/PX4-Bootloader #181: V5 Bootloader not working on newer Macs

ryanjAA, 2021-01-02:
Newer Macs is firstly subjective but in the tests I did, as I rolled back the hardware and software of the Mac computers being used it started working again.

Problem: The V5 bootloader doesn't recognize the board from QGC when trying to flash PX4 (or Ardupilot) firmware. I tried with 2x MRO Pixhawk 1, 1x Generic Pixhawk 1, 5+ cables, 3x computers and several USB-C adapters.

When on bootloader v4 there were no problems on any of the machines. The issue becomes that you can't upload fmu_v3 firmware unless upgrading the bootloader (unless doing it manually) so when flashing the bootloader (all done via QGC 4.0.11 and PX4 v1.11.2), the ability to upgrade the firmware afterwards from QGC is no longer possible. It does not recognize the chip size, just hangs at Found Device: PX4 FMU V2 9 (see image). It also doesn't give the connected to boot loader on the next line (I tried QGC daily as of today and it also did not work).

Building and uploading the firmware from command line does work but it does ask 5-7 times to remove the USB (which I didn't do, just left it) and then it will flash without issue. 

If I load the Chibios boot loader on the Pixhawks any of the computers will pick them up instantly for upgrading the firmware (PX4 works fine this way), only when putting the v5 firmware back does it stop working.

This is where the newer Macs get to be part of the problem. On macOS Big Sur and Catalina (and usb-c) Macs, the problem above is present 99% of the time. It's not 100% because after repeatedly doing it, occasionally (rarely) I could get it to show up but this was 1-2%.

Going back to an older Mac with High Sierra and Mojave and all worked (or something to do with USB-C?)

The current fix for the time being is to get the Chibios loaded up and just leave it on there. No issues with that on any machines but obviously not ideal.

It's strange and I don't have it nailed down to something exact but it is somewhere lurking in the above but I would imagine it will start to effect more people.

I'd rule out hardware as I did this across multiple machines and units and cables (and versions of QGC) and making the switch to a different bootloader does fix it.

[image]
[image]
[Screen Shot 2021-01-01 at 11.49.16 AM copy.pdf](https://github.com/PX4/PX4-Bootloader/files/5759752/Screen.Shot.2021-01-01.at.11.49.16.AM.copy.pdf)

LorenzMeier, 2021-01-10:
Thanks for reporting! I'm looking into this over the next few days.

mrpollo, 2021-04-28:
Hey @ryanjAA have you tried flashing the PX4 bootloader on your board?

I found a similar issue on Slack from Jonas

https://px4.slack.com/archives/C0V533X4N/p1618609738321100

https://docs.px4.io/master/en/advanced_config/bootloader_update.html#qgroundcontrol-bootloader-update

MaEtUgR, 2021-04-28:
Related context by @davids5:
There was an issue with the bootloader in the past. If you have the bootloader from the broken window this could be the issue.
Sequence:
Added sequencing Dec 12, 2019
https://github.com/PX4/PX4-Bootloader/commit/d1cd19871abede73f8f7bd1017a5b8d46cd3023f
<
<  You would have the problem
<
Fixed QGC  Aug 14, 2020 https://github.com/PX4/PX4-Bootloader/commit/b2acae83
Fixed the px4io Aug 31, 2020 https://github.com/PX4/PX4-Bootloader/commit/0b46110ad6c1bb1cc5660a7381a51050a114a8ca

Summary: If you see this issue please try updating the bootloader: https://docs.px4.io/master/en/advanced_config/bootloader_update.html#qgroundcontrol-bootloader-update

ryanjAA, 2021-04-28:
I’ll try again but I could only get it working with the chibios bootloader. I did try 4-5 different Pixhawk 1 and uploading the PX4  bootloader but it was the same every time. Then putting the chibios on it and it was fine. 

I did however get it to connect every 20 times or so for one time but generally it didn’t.

julianoes, 2021-05-04:
I just tried this as well, here are my results:

- MacMini 8,1, macOS 11.2.3, using the USB-A ports
- tested using Pixracer and Pixhawk 4, both using bootloaders flashed yesterday
- Flashing in latest QGC worked fine
- Flashing using px_uploader.py worked fine

Are you on a new M1 and that's why it's different for you?