PX4/PX4-Bootloader #182: which file should I use when I use a stm32h7 board？

Dunkem, 2021-01-10:
hi，I am fresher of px4，and now I have a board（stm32h7），but I didn't find any bootloader file suite this kind of board? So, can you tell me which file I could down for my stm32h
7 board?
Thank's a lot!

LorenzMeier, 2021-01-10:
The bootloader for H7 (to the Pixhawk FMUv6X standard) is here: https://github.com/PX4/PX4-Autopilot/tree/master/boards/px4/fmu-v6x

LorenzMeier, 2021-01-10:
Use the standard PX4 build environment (http://docs.px4.io/master/en/development/development.html)

and then run:

```make px4_fmu-v6x_bootloader```

It will create the bootloader ELF file in `PX4-Autopilot/build/px4_fmu-v6x_bootloader/px4_fmu-v6x_bootloader.elf` that you can flash via SWD or DFU.

closed 2021-01-10 by LorenzMeier

Dunkem, 2021-01-10:
OK，thanks for your help, can I use the hex file or only the bin and elf file are ok for the board?

发自我的iPhone

 

 
 
 
It will create the bootloader ELF file PX4-Autopilot/build/px4_fmu-v6x_bootloader/px4_fmu-v6x_bootloader.elf that you can flash via SWD or DFU.

LorenzMeier, 2021-01-10:
An ELF file always contains a HEX: Just copy the output: ```arm-none-eabi-objcopy -O ihex px4_fmu-v6x_bootloader.elf px4_fmu-v6x_bootloader.hex```

That will give you px4_fmu-v6x_bootloader.elf px4_fmu-v6x_bootloader.hex

Dunkem, 2021-01-10:
Fine！thanks for your reply！

发自我的iPhone

LorenzMeier, 2021-01-10:
Also documented here now: https://github.com/PX4/PX4-user_guide/blob/master/en/advanced_config/bootloader_update.md

Dunkem, 2021-01-10:
OK，I will read it and start my way of flight，and also send my sincere appreciation to you!

发自我的iPhone

Dunkem, 2021-01-12:
Hi, maybe I will brother you again, after download the bootloader file (a hex file via st link),I connect my stm32 to computer via Type-C port, but I get a "USB device not recognized" error. Is there any driver I need to install before flash firmware?
Thanks again for your help!

发自我的iPhone

LorenzMeier, 2021-01-12:
Please install QGroundControl to try.

Dunkem, 2021-01-12:
I download this software, but it also can't be recognized by QGroundControl

发自我的iPhone

Dunkem, 2021-01-12:
and I test the PE5's voltage, it's zero. It seems the bootloader doesn't run.this image show what I am using.
Thanks for your help!

发自我的iPhone

LorenzMeier, 2021-01-12:
Your board would need to follow the FMUv6X Pixhawk design - it is not sufficient that it is just a H7. Does it follow the v6X pinout?

Dunkem, 2021-01-12:
It is a core board just with a usb port. this board's type is stm32H743VIT6, and I just lead out the pin of it, and the type-c port is PA11(USB_D_N) and PA12(USB_D_P). 

发自我的iPhone

Dunkem, 2021-01-14:
Hi, Lorenz may I bother you again. I find my problem may be my board Crystal oscillator which is 8Mhz is different with bootloader. Can you tell me where to modify the Crystal oscillator? Thanks again for your help!

发自我的iPhone

Dunkem, 2021-01-18:
Hi, Lorenz. I am a member of FAST Lab(http://zju-fast.com/). And all our works will be open source. I am so glad that you can help me build bootloader file for my board. I am really fresher about hardware and px4 firmware. And now we want to design a small drone controlled by px4 and only with a IMU. So the first step we want to test the stability of the controller, but I failed to flash bootloader to the board. The problem is I download bootloader to my board, but the usb port doesn't be recognized by my computer so I can't download firmware to the board. It seems the bootloader doesn't be launched. Can you help me which file should I modify for this problem. The attachment is our board hardware design. And we also will show our acknowledgement to you and your team in our paper.
Thanks for your time to read this email.

Dunkem, 2021-01-19:
Lorenz, I am so sorry for disturb you and I should make appoloies to what I've said yestoday. I am so happy find this open source project, and you and your team make a great example for me. And I know this problem should be solved by myself. Please forget what I've done for you, it may make you feel unhappy. I say sorry again.
Have a nice day! Thanks a lot for you and your team.
Best wishes!

MadhavParashar12, 2024-10-29:
Hii,
hi, I am trying to install px4 autopilot on STM32h&57I-eval, I believe you know how to do it, please help.

liuyuan018, 2025-04-16:
您好，我也是 px4 的新手，现在我手里的板子（stm32h743VIT6），但是我按照上边烧录H7 的引导程序（符合 Pixhawk FMUv6X 标准）之后，电脑并没有识别出来STM32H743VIT6板子，QGC也识别不出来这个板子。请问哪里出错了？该怎么解决呢？
非常感谢！

MadhavParashar12, 2025-05-21:
hi,
I am also a novice and am still finding a solution.