PX4/PX4-Bootloader #196: PX4 v1.9.0 NOT GIVING SERVO OUTPUTS ON Cuav5+

vineethtom, 2021-04-15:
No servo outputs and Starting PX4IO beep continously while flashing pixhawk v1.9.0.

Tried using V1.11 which works perfectly.
Since I am working on fixed wings and also need a HiL environment for the same, I have to stick on to V1.9.0.

Could someone help me ASAP.

my mail id is vineethinside@gmail.com.

Its bugging my head for a long time now

I am relatively new to pixhawk.

LorenzMeier, 2021-04-15:
CUAV v5 is not really supported in 1.9 - are you sure that HIL is not working in 1.11? If so, please file an issue in PX-Autopilot, not here

closed 2021-04-15 by LorenzMeier

vineethtom, 2021-04-16:
So are you saying that V1.9.0 would work on CuAV v5+?

Hil support for fixed wings in 1.11.0 is not there.

vineethtom, 2021-04-16:
Correction: Are you saying 1.9.0 doesnt work on Cuav v5+?

LorenzMeier, 2021-04-16:
Please comment on this issue and share what you're exactly seeing. HITL is supported for fixed wing aircraft with autostart ID 1000, so I'm not quite sure where it breaks for you.

https://github.com/PX4/PX4-Autopilot/issues/17410

I will not continue to respond here, because this is the bootloader repository.