PX4/PX4-Bootloader #216: Question: Is secure boot working stably?

divyanshu1234, 2022-04-08:
Hello,
We are planning to use this on one of our drones (fmuv5x). I can see that the last release was 6 years ago. Since then some important features like secure boot have been implemented. Therefore, I wanted to know if the master build of the bootloader is working stably for fmuv5x.

PS.
Please provide any documentation on how to use this.

dagar, 2022-04-08:
@jlaitine any comment?

jlaitine, 2022-04-08:
I didn't check the latest master, my own branch is from March 2021. There doesn't seem to be much differencies, however, so it should be fine with the master. Please let me know if you have any problems with it, I can check it out.

To use it, you need to 1) enable signature check and embed the public key to the bootloader binary.  2) Add the TOC structure to px4 firmare and sign the binary. 3) Protect the public key embedded in the bootloader from tampering

1) The procedure for the bootloader is:

$ export CRYPTO_HAL=monocypher
$ export PUBLIC_KEY0=/build/crypto_hal/test_key/key0.pub
$ make px4fmuv5x_bl

To use your own keys, just point the PUBLIC_KEY0 to your own key-pair public part. But I suggest first testing with the provided test keys.

After flashing the bootloader you'll find out that the board doesn't boot any more

2) The procedure for px4 firmare is as follows

- Enable TOC to the firmare. Here is an example:

 https://github.com/tiiuae/px4-firmware/commit/395d92165810d43ddcac4108dd59b9ba3c658cd4

- Build the firmware using signing tool

$ export SIGNING_TOOL=./Tools/cryptotools.py
$ make px4_fmu-v5x_default

The python script uses the same test key by default as what was used by default for the bootloader above. Again, to use your own key pair, the private key can be given as an argument to the "cryptotools.py" (via an envronment variable), or alternatively the whole script can be changed to your own one, which e.g. can operate over in your own infrastructure signing server or whatever. Just check the python script and what the build script gives to it as arguments.

3) To protect the public key

The 1) and 2) are enough to test whether the signature check works and for the R&D. If you want to use this for a product, you need to enable RDP protection (level 2) on the STM chip to protect the bootloader. This can be done via JTAG and st-link utility. Don't do this until you are sure that everything is ok, it permanenty disables JTAG etc. Please note that I am *not* promising that the STM RDP protection is unbreakable ;) It is. however, commonly used in commercial products.

Please let me know if you need any more assistance!

divyanshu1234, 2022-04-09:
Thanks a lot @jlaitine. I'll try it out in a few days. If things go well, I'll try to create a PR of the step by step process.

Prishat, 2023-10-12:
Has anyone tried building the bootloader and firmware for pix32 v6 (_px4_fmu-v6c_) microcontroller? 
I have tried the above mentioned process on v5x and it is working as expected. Please provide the documentation if anyone was able to successfully do it.